intel/ros2_openvino_toolkit #3: request: ROS1 wrapper

gbr1, 2018-10-16:
Why is it only available for ROS2 and not ROS1?

LewisLiuPub, 2018-10-17:
@gbr1 
thank you for your attention.
Actually, porting to ROS1 has been in our development plan. It is supposed to be released in the soon future.

By the way, I am happy to know if some packages on which you are working will depend on the similar features of ros2_openvino_toolkit?

gbr1, 2018-10-17:
@LewisLiuPub I'm working on Erwhi project,  robotic kit by AAEON based on Up2, Intel Realsense depth D435 and AiCore (Movidus).

Link to project: 
https://gbr1.github.io/erwhi_hedgehog.html

Obviously due critical releases and software unstability of ROS2, it is based on ROS kinetic/melodic.

Now I'm using old ncs sdk with old ncappzoo and bad ros node Intel Ros Movidius, to integrate caffe in ros using offload on AIcore.  Ubfortunately it is really old and bad written (no ros standard topic published).
So I'm looking for people who already started to develop openvino node for ros.

Do you know when is OpenVino Ros1 package going to be released?

Thanks!

LewisLiuPub, 2018-10-17:
@gbr1 
are you aware of another ros package (https://github.com/intel/ros_intel_movidius_ncs) supporting ncs sdk and ros standard topic publishing? If you haven't touched this package, it's worthy to be tried in your project.
I have no detailed release schedule just now. To confidently say, it will be released in this quarter. ^,^. 

Not sure if it would be the 3rd solution: would you or other guys here like to be involved into the porting tasks, in order to get a quicker progress? ^;^

gbr1, 2018-10-17:
yes, that one is the orrible node xD

It's awesome to know that it is in this quarter.

I don't know how I can help you.

Maybe you can contact me in private by email: giovannididio.bruno@gmail.com

💪🔥

LewisLiuPub, 2018-11-15:
@gbr1 thank you for your consideration and waiting.
We have been putting efforts into porting it to ROS system. we are trying our best to fasten the progress of external release.

LewisLiuPub, 2018-12-10:
ROS version of openVINO toolkit has been already released [here](https://github.com/intel/ros_openvino_toolkit), so close this bug now.
ROS version will be continuously updated with new features and bug-fixing.

closed 2018-12-10 by LewisLiuPub