intel/ros2_openvino_toolkit #22: After get openvino R4 open source code, colcon build occurred error

RachelRen05, 2018-11-28:
After get openvino R4 open source code, **colcon build** occurred error:
`--- stderr: dynamic_vino_lib                                                                                
In file included from /home/iei/ros2_overlay_ws/src/ros2_openvino_toolkit/dynamic_vino_lib/src/factory.cpp:25:0:
/home/iei/ros2_overlay_ws/src/ros2_openvino_toolkit/dynamic_vino_lib/include/dynamic_vino_lib/factory.hpp:34:43: fatal error: mkldnn/mkldnn_extension_ptr.hpp: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/dynamic_vino_lib.dir/src/factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
`

LewisLiuPub, 2018-12-12:
@RachelRen05 I knew you were focusing on this bug. let me know your progress, thanks a lot.

LewisLiuPub, 2018-12-12:
fixed in [PR27](https://github.com/intel/ros2_openvino_toolkit/pull/27).

closed 2018-12-12 by LewisLiuPub