intel/ros2_openvino_toolkit #58: when launching face detection, there was no result in /ros2_openvino_toolkit/face_detection topic

RachelRen05, 2019-03-01:
when launching face detection,

` ros2 launch dynamic_vino_sample pipeline_people_oss.launch.py`

there was no result in /ros2_openvino_toolkit/face_detection topic.

`ros2 topic echo /ros2_openvino_toolkit/face_detection`

the result is in object detection topic:

`ros2 topic echo /openvino_toolkit/detected_objects`

EwingKang, 2019-04-16:
I can confirm that there's no topic on `/ros2_openvino_toolkit/face_detection`
However, all the other topics under the same namespace: 
```
/ros2_openvino_toolkit/age_genders_Recognition
/ros2_openvino_toolkit/emotions_recognition
/ros2_openvino_toolkit/headposes_estimation
/ros2_openvino_toolkit/image_rviz
```
are all working normally.

houk-intel, 2019-04-23:
I suggest we remove the _/ros2_openvino_toolkit/face_detection_ topic and categorize it into ObjectDetection, for the two reasons below.

1. We have **object_name** in ObjectDetection RosTopic output, which can help distinguish the different types of objects.
2. We also have people detection, vehicle detection, and they are under the /openvino_toolkit/detected_objects topic. Face detection should not be a special case.

LewisLiuPub, 2019-04-24:
I agree with @houk-intel, face is just one of the objects detected, so there is no need to make it special. Furthermore, it would be easier if we consider of "detection filtering" feature.

LewisLiuPub, 2023-04-24:
Close. Face Detection is changed as one implementation of Object Detection.