intel/ros2_openvino_toolkit #84: people_msgs build failed when using lastest  ros2_core(20190510).

congliu0913, 2019-05-10:
The error log is shown below:
```
[Processing: object_analytics_msgs, people_msgs]                                                                  
--- stderr: people_msgs                                                                                            
In file included from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/AgeGender_.h:15:0,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/ccpp_AgeGender_.h:4,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/age_gender__rosidl_typesupport_opensplice_cpp.hpp:14,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/age_gender__type_support.cpp:19:
/home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/msg/dds_opensplice/AgeGenderStamped_.h:30:79: error: type/value mismatch at argument 1 in template parameter list for ‘template<class T, class X> class DDS_DCPSUVLSeq’
                typedef DDS_DCPSUVLSeq < AgeGender_, struct _objects__seq_uniq_> _objects__seq;
                                                                               ^
/home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/msg/dds_opensplice/AgeGenderStamped_.h:30:79: note:   expected a type, got ‘AgeGender_’
In file included from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/Emotion_.h:15:0,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/ccpp_Emotion_.h:4,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/emotion__rosidl_typesupport_opensplice_cpp.hpp:14,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/emotion__type_support.cpp:19:
/home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/msg/dds_opensplice/EmotionsStamped_.h:30:78: error: type/value mismatch at argument 1 in template parameter list for ‘template<class T, class X> class DDS_DCPSUVLSeq’
                typedef DDS_DCPSUVLSeq < Emotion_, struct _emotions__seq_uniq_> _emotions__seq;
                                                                              ^
/home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/msg/dds_opensplice/EmotionsStamped_.h:30:78: note:   expected a type, got ‘Emotion_’
In file included from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/HeadPose_.h:15:0,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/ccpp_HeadPose_.h:4,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/head_pose__rosidl_typesupport_opensplice_cpp.hpp:14,
                 from /home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/head_pose__type_support.cpp:19:
/home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/msg/dds_opensplice/HeadPoseStamped_.h:30:80: error: type/value mismatch at argument 1 in template parameter list for ‘template<class T, class X> class DDS_DCPSUFLSeq’
                typedef DDS_DCPSUFLSeq < HeadPose_, struct _headposes__seq_uniq_> _headposes__seq;
                                                                                ^
/home/intel/robot_sdk/sdk_ws/modules_ws/build/people_msgs/rosidl_typesupport_opensplice_cpp/people_msgs/msg/dds_opensplice/HeadPoseStamped_.h:30:80: note:   expected a type, got ‘HeadPose_’
make[2]: *** [CMakeFiles/people_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/age_gender__type_support.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/people_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/emotion__type_support.cpp.o] Error 1
make[2]: *** [CMakeFiles/people_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/people_msgs/srv/dds_opensplice/head_pose__type_support.cpp.o] Error 1
make[1]: *** [CMakeFiles/people_msgs__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< people_msgs	[ Exited with code 2 ]

```

testkit, 2019-07-11:
@chaoli2 @LewisLiuPub , what's the solution for this bug? Did we apply any workaround before opensplice has the real fix?

chaoli2, 2019-07-17:
@testkit The root cause of this issue is that topic name and service name are not allowed to be duplicated under opensplice. Re-name the topic name or service name will fix this issue temporarily.

LewisLiuPub, 2019-09-27:
fixed by #97

closed 2019-09-27 by LewisLiuPub