intel/ros2_openvino_toolkit #88: build failed when building dynamic_vino_lib

xhuan28, 2019-05-23:
$colcon build --symlink-install

--- stderr: dynamic_vino_lib                           
CMake Error at CMakeLists.txt:68 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< dynamic_vino_lib	[ Exited with code 2 ]

xhuan28, 2019-05-23:
This issue is caused by not supporting 2019R1 release of OpenVINO toolkit

LewisLiuPub, 2019-09-27:
Close, because it can't be reproduced in the latest version.

closed 2019-09-27 by LewisLiuPub

portgasray, 2019-11-22:
#160

portgasray, 2019-11-22:
> This issue is caused by not supporting 2019R1 release of OpenVINO toolkit

so, what's your suggestion?

I used your doc [OPEN_SOURCE_CODE_README](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md), and tried the following the step showing as the pic, it still happened like above!  Check my issue #160 

[image]