intel/ros2_openvino_toolkit #106: run multiple pipeline failed

RachelRen05, 2019-07-15:
When I tried to launch multiple pipeline with StandardCamera and RealSenseCamera, the RealSenseCamera works fine, but the StandardCamera doesn't work.

luyang00, 2019-07-15:
This problem mainly because when plugin Realsense camera, it will automatically mount three camera devices e.g. /dev/video0 /dev/video1 /dev/video2. Previous initialization based on opencv videio API cannot detect one is a webcamera or a realsense camera. 
In this PR, I use v4l2 api to detect standard camera during initalization.

luyang00, 2019-07-15:
Fix in https://github.com/intel/ros2_openvino_toolkit/pull/107

LewisLiuPub, 2019-09-27:
#107 fixed this issue under Linux environment.
I am going to keep this issue open for a while, in case the possibilities from the community of fixing it in Windows/Mac Env. :)

LewisLiuPub, 2023-04-24:
Sounds the better solution is:
- add one item "input-device" in YAML configuration file when 'inputs" is set to "StandardCamera".
- Update "openvino_param_lib" to support parsing of the new item.
- Update "Input::StandardCamera" class to support the new configuration.

tmcourie, 2023-04-26:
ROS Team 1 working on this

ChloeWu1, 2023-04-29:
ROS_Teams2 working on this issue.

ChloeWu1, 2023-04-29:
Please review PR #293

gitpqLee, 2025-02-20:
Close due to PR merged,

closed 2025-02-20 by LewisLiuPub