intel/ros2_openvino_toolkit #119: why need link src after colcon build

yechun1, 2019-09-23:
from the build package Guide, after colcon build, there is one command to link ros2_openvino_toolkit to /opt/openvino_toolkit, don't know what is the purpose, could this soft link be removed?

"sudo ln -sf ~/ros2_overlay_ws/src/ros2_openvino_toolkit /opt/openvino_toolkit/"

LewisLiuPub, 2019-09-27:
Yes, you are right.
It is not good to link a source code folder to a system folder. It would be better to change the solution this way:
1. Parameter file (.yaml) uses relative paths for the needed file location. 
2. Launch.py file can pass the prefix of the absolute path to the openvino sample program.
3. In the program, generate the absolute path by combining the prefix and relative paths in parameter file.

yechun1, 2019-11-11:
I found there are different data file point to different folder.
```

sample/tests/testParam/param/image_object_service_test.yaml:18:  input_path: "/home/intel/ros2_overlay_ws/src/ros2_openvino_toolkit/data/images/car_vihecle.png"
sample/tests/testParam/param/pipeline_image_test.yaml:4:  input_path: /opt/openvino_toolkit/ros2_openvino_toolkit/data/images/team.jpg
```

Could we put all the data as resources of ros2_openvino_package, and use relative in yaml?
for example :
```
Input_path: "data/images/car_vihecle.png"
 input_path: "data/images/team.jpg"
```

Could we use ros2 API to get data path like below?
```
#include <ament_index_cpp/get_package_share_directory.hpp>
// may throw PackageNotFoundError exception
std::string package_share_directory = ament_index_cpp::get_package_share_directory("my_package_name");
```
Here is a reference :https://answers.ros.org/question/288501/ros2-equivalent-of-rospackagegetpath/

closed 2026-01-12 by jb-balaji