intel/ros2_openvino_toolkit #123: [robot_devkit]Is there a better way to deal with the situation that label is null?

xhuan28, 2019-10-11:
When I compose multiple nodes into a container process through a launch file, "label:null" in the yaml file will cause the issue as follows:

> Future exception was never retrieved
future: <Future finished exception=TypeError('Unexpected type for parameter value None',)>
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/intel/workspace/ros2_ws/build/launch_ros/launch_ros/actions/load_composable_nodes.py", line 151, in _load_in_sequence
    self._load_node(next_composable_node_description, context)
  File "/home/intel/workspace/ros2_ws/build/launch_ros/launch_ros/actions/load_composable_nodes.py", line 114, in _load_node
    context, composable_node_description.parameters
  File "/home/intel/workspace/ros2_ws/build/launch_ros/launch_ros/utilities/to_parameters_list.py", line 49, in to_parameters_list
    context, normalize_parameter_dict(yaml.load(f))
  File "/home/intel/workspace/ros2_ws/build/launch_ros/launch_ros/utilities/normalize_parameters.py", line 158, in normalize_parameter_dict
    raise TypeError('Unexpected type for parameter value {}'.format(repr(value)))

chaoli2, 2019-10-15:
I have submitted a [PR](https://github.com/intel/ros2_openvino_toolkit/pull/125) to resolve this issue.

xhuan28, 2019-10-17:
Thanks. I think this issue can be closed.

closed 2019-10-17 by xhuan28