intel/ros2_openvino_toolkit #127: Installation guide Issue

ahuizxc, 2019-10-17:
doc/BINARY_VERSION_README.md
at 4.Building and Installation
after cd /opt/intel/openvino/deployment_tools/inference_engine/samples/
I recommand run ./build_samples.sh instead of manually build the samples
and then:
 export CPU_EXTENSION_LIB=$HOME/inference_engine_samples_build/intel64/Release/lib/libcpu_extension.so
 export GFLAGS_LIB=$HOME/inference_engine_samples_build/intel64/Release/lib/libgflags_nothreads.a

LewisLiuPub, 2019-10-18:
thank you very much for your good suggestion, Hui.
I am going to think of the necessities to update.

portgasray, 2019-11-22:
Keep watching

portgasray, 2019-11-22:
Also, I want to know the code in this CMakeLists  [line75](https://github.com/intel/ros2_openvino_toolkit/blob/a24c0e071de72fd8a58a4bdea3f49c261d627b4f/dynamic_vino_lib/CMakeLists.txt#L75) , how does it get the cpu_extension? 

should I export in the environment first?

ahuizxc, 2019-11-22:
> Also, I want to know the code in this CMakeLists [line75](https://github.com/intel/ros2_openvino_toolkit/blob/a24c0e071de72fd8a58a4bdea3f49c261d627b4f/dynamic_vino_lib/CMakeLists.txt#L75) , how does it get the cpu_extension?
> 
> should I export in the environment first?
> 
> By the way, although I followed the [OPEN_SOURCE_CODE_README](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md), but I didn't execute the `Install OpenCL Driver for GPU`. Is this only support for **Intel GPU**, Does this have any effect if I didn't install it?

you need to execute `Install OpenCL Driver for GPU` and then do
``bash
export InferenceEngine_DIR=/opt/openvino_toolkit/dldt/inference-engine/build/
 export CPU_EXTENSION_LIB=/opt/openvino_toolkit/dldt/inference-engine/bin/intel64/Release/lib/libcpu_extension.so
 export GFLAGS_LIB=/opt/openvino_toolkit/dldt/inference-engine/bin/intel64/Release/lib/libgflags_nothreads.a
```
on your terminal when you execute `colcon build`,otherwise you may faild to install this package.
By the way, this project is only support Intel GPU:) you must install the OpenCL Driver for GPU.

portgasray, 2019-11-22:
> > Also, I want to know the code in this CMakeLists [line75](https://github.com/intel/ros2_openvino_toolkit/blob/a24c0e071de72fd8a58a4bdea3f49c261d627b4f/dynamic_vino_lib/CMakeLists.txt#L75) , how does it get the cpu_extension?
> > should I export in the environment first?
> > By the way, although I followed the [OPEN_SOURCE_CODE_README](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md), but I didn't execute the `Install OpenCL Driver for GPU`. Is this only support for **Intel GPU**, Does this have any effect if I didn't install it?
> 
> you need to execute `Install OpenCL Driver for GPU` and then do
> 
> ```shell
> export InferenceEngine_DIR=/opt/openvino_toolkit/dldt/inference-engine/build/
>  export CPU_EXTENSION_LIB=/opt/openvino_toolkit/dldt/inference-engine/bin/intel64/Release/lib/libcpu_extension.so
>  export GFLAGS_LIB=/opt/openvino_toolkit/dldt/inference-engine/bin/intel64/Release/lib/libgflags_nothreads.a
> ```
> 
> on your terminal when you execute `colcon build`,otherwise you may faild to install this package.
> By the way, this project is only support Intel GPU:) you must install the OpenCL Driver for GPU.

Yeah, agree with u. I met with exception(shown below) when I run `ros2 run grasp_ros2 grasp_ros2 __params:=src/ros2_grasp_library/grasp_ros2/cfg/grasp_ros2_params.yaml` under the ros2_grasp_library

**Exception**
```
terminate called after throwing an instance of 'InferenceEngine::details::InferenceEngineException'
  what():  Cannot find plugin to use :Tried load plugin : clDNNPlugin,  error: Plugin clDNNPlugin cannot be loaded: cannot load plugin: clDNNPlugin from ../../../lib/intel64: Cannot load library '../../../lib/intel64/libclDNNPlugin.so': ../../../lib/intel64/libclDNNPlugin.so: cannot open shared object file: No such file or directory, skipping
cannot load plugin: clDNNPlugin from : No OpenCL device found which would match provided configuration:
    GeForce GTX 1080 Ti: invalid vendor type
    GeForce GTX 1080 Ti: invalid vendor type, skipping
```
I guessed this due to lack of  opecl drive .

portgasray, 2019-11-22:
oh, I tried to install your OpenCL driver, then I still can not run grasp code,  I am a little disappointed.

ahuizxc, 2019-11-22:
> oh, I tried to install your OpenCL driver, then I still can not run grasp code, I am a little disappointed.

maybe in your configure file, you set your GTX1080 for openvino use, but openvino doesn't support nvidia GPU so unload your GTX1080 and try again:), don't give up~~

portgasray, 2019-11-22:
> > oh, I tried to install your OpenCL driver, then I still can not run grasp code, I am a little disappointed.
> 
> maybe in your configure file, you set your GTX1080 for openvino use, but openvino doesn't support nvidia GPU so unload your GTX1080 and try again:), don't give up~~

thank you for your encouragement, I am trying to solve this problem by setting the device model to  `CPU`

ahuizxc, 2019-11-23:
> thank you for your encouragement, I am trying to solve this problem by setting the device model to `CPU`

glad to hear this my friend!

LewisLiuPub, 2019-11-25:
Thank you, @ahuizxc , very much for helping to watch on this project!
Thank you, @portgasray for paying attention to this project, may you have a good trip.  @,@!

closed 2019-11-25 by LewisLiuPub