intel/ros2_openvino_toolkit #132: not found setprecision while compile dynamic_vino_lib

yechun1, 2019-10-24:
Openvino build error while build https/github/intel/robot_devkit.git.
It look the file miss the include <iomanip>

> --- stderr: dynamic_vino_lib                                                                                                     
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp: In member function ‘virtual void Outputs::ImageWindowOutput::accept(const std::vector<dynamic_vino_lib::ObjectSegmentationResult>&)’:
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:178:44: error: ‘setprecision’ is not a member of ‘std’
       ostream << "[" << std::fixed << std::setprecision(3) << fd_conf << "]";
                                            ^~~~~~~~~~~~
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:178:44: note: suggested alternative: ‘streamsize’
       ostream << "[" << std::fixed << std::setprecision(3) << fd_conf << "]";
                                            ^~~~~~~~~~~~
                                            streamsize
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp: In member function ‘virtual void Outputs::ImageWindowOutput::accept(const std::vector<dynamic_vino_lib::FaceDetectionResult>&)’:
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:197:44: error: ‘setprecision’ is not a member of ‘std’
       ostream << "[" << std::fixed << std::setprecision(3) << fd_conf << "]";
                                            ^~~~~~~~~~~~
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:197:44: note: suggested alternative: ‘streamsize’
       ostream << "[" << std::fixed << std::setprecision(3) << fd_conf << "]";
                                            ^~~~~~~~~~~~
                                            streamsize
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp: In member function ‘virtual void Outputs::ImageWindowOutput::accept(const std::vector<dynamic_vino_lib::ObjectDetectionResult>&)’:
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:213:44: error: ‘setprecision’ is not a member of ‘std’
       ostream << "[" << std::fixed << std::setprecision(3) << fd_conf << "]";
                                            ^~~~~~~~~~~~
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:213:44: note: suggested alternative: ‘streamsize’
       ostream << "[" << std::fixed << std::setprecision(3) << fd_conf << "]";
                                            ^~~~~~~~~~~~
                                            streamsize
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp: In member function ‘virtual void Outputs::ImageWindowOutput::accept(const std::vector<dynamic_vino_lib::AgeGenderResult>&)’:
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:242:43: error: ‘setprecision’ is not a member of ‘std’
     ostream << "[Y" << std::fixed << std::setprecision(0) << results[i].getAge() << "]";
                                           ^~~~~~~~~~~~
/home/rdktest/robot_devkit/rdk_ws/perception_ws/src/intel/ros2_openvino_toolkit/dynamic_vino_lib/src/outputs/image_window_output.cpp:242:43: note: suggested alternative: ‘streamsize’
     ostream << "[Y" << std::fixed << std::setprecision(0) << results[i].getAge() << "]";
                                           ^~~~~~~~~~~~
                                           streamsize
make[2]: *** [CMakeFiles/dynamic_vino_lib.dir/src/outputs/image_window_output.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/dynamic_vino_lib.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< dynamic_vino_lib	[ Exited with code 2 ]

yechun1, 2019-10-24:
PR to fix this issue: https://github.com/intel/ros2_openvino_toolkit/pull/133

LewisLiuPub, 2019-10-24:
Thanks a lot for your finding and fixing. :)