intel/ros2_openvino_toolkit #155: ros2_overlay_ws colcon build --symlink-install error

arun0303, 2019-11-14:
During step 4. Building and Installation
-->Install ROS2_OpenVINO packages
-->Build package
cd ~/ros2_overlay_ws
colcon build --symlink-install
 The following error 
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libboost_python3.so', needed by 'src/libcv_bridge.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/CMakeFiles/cv_bridge.dir/all] Error 2
make: *** [all] Error 2

portgasray, 2019-11-22:
> During step 4. Building and Installation
> -->Install ROS2_OpenVINO packages
> -->Build package
> cd ~/ros2_overlay_ws
> colcon build --symlink-install
> 
> ## [Processing: cv_bridge, message_filters, object_msgs, realsense_msgs]
> --- stderr: cv_bridge
> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libboost_python3.so', needed by 'src/libcv_bridge.so'. Stop.
> make[2]: *** Waiting for unfinished jobs....
> make[1]: *** [src/CMakeFiles/cv_bridge.dir/all] Error 2
> make: *** [all] Error 2
> Failed <<< cv_bridge [ Exited with code 2 ]
> Aborted <<< realsense_msgs
> Aborted <<< message_filters
> Aborted <<< object_msgs
> 
> Summary: 0 packages finished [2min 4s]
> 1 package failed: cv_bridge
> 3 packages aborted: message_filters object_msgs realsense_msgs
> 1 package had stderr output: cv_bridge
> 15 packages not processed

u should go to the repo [ros2_intel_realsense](https://github.com/intel/ros2_intel_realsense), and see the README, to install the dependencies, such as
```
sudo apt-get install ros-dashing-cv-bridge ros-dashing-librealsense2 ros-dashing-message-filters ros-dashing-image-transport
```

Check that! Good Luck!

arun0303, 2019-11-25:
> u should go to the repo [ros2_intel_realsense](https://github.com/intel/ros2_intel_realsense), and see the README, to install the dependencies, such as
>
> ```
> sudo apt-get install [...]

Yeah those dependencies were installed previously and tried again also but no effect. pls see the screenshot.
[image]

portgasray, 2019-11-27:
> Yeah those dependencies were installed previously and tried again also but no effect. pls see the screenshot.
> [image]

`sudo apt remove ros-dashing-librealsense2`, give it a shot! (I am not sure if this will work!)

then I will use `rm install log build -rf` and `colcon build --symlink-install` to reproduce the error under _ros2 workspace_.

arun0303, 2019-11-28:
> `sudo apt remove ros-dashing-librealsense2`, give it a shot! (I am not sure if this will work!)
> 
> then I will use `rm install log build -rf` and `colcon build --symlink-install` to reproduce the error under _ros2 workspace_.

It did not worked.
[image]

portgasray, 2019-12-02:
> > `sudo apt remove ros-dashing-librealsense2`, give it a shot! (I am not sure if this will work!)
> > then I will use `rm install log build -rf` and `colcon build --symlink-install` to reproduce the error under _ros2 workspace_.
> 
> It did not worked.

since your error caused by boost, check whether installed correctly!

arun0303, 2019-12-02:
> since your error caused by boost, check whether installed correctly!
I had verified boost is installed properly. pls find the screenshot
[image]

portgasray, 2019-12-03:
see the error information: `needed by 'src/libcv_bridge.so'. Stop.` 
there is no` libcv_bridge.so` in my `ros2_overlay_ws/src` , so if you can show me where the `libcv_bridge.so` file

arun0303, 2019-12-03:
libcv_bridge.so  file is not available in my ros2_overlay_ws/src directory and its available inside /opt/ros/dashing/lib directory.

portgasray, 2019-12-03:
> libcv_bridge.so file is not available in my ros2_overlay_ws/src directory and its available inside /opt/ros/dashing/lib directory.

more information about the `ros2_overlay_ws/src/vision_opencv`:
* git branch

portgasray, 2019-12-03:
yeah, you should check the CMakeLists under `cv_bridge`, you'll find the valid information: 
```
      # This is a bit of a hack to suppress a warning
      #   No header defined for python3; skipping header check
      # Which should only affect Boost versions < 1.67
      # Resolution for newer versions:
      #  https://gitlab.kitware.com/cmake/cmake/issues/16391
```
the sub-links in the [issuse](https://gitlab.kitware.com/cmake/cmake/issues/16391) are very useful, especially the last one, 
[image]

you should view it carefully and  according to my experience, try the following:
update [line30 ](https://github.com/ros-perception/vision_opencv/blob/b19d3d10c0c822c454047c1c4950ab22dc4a9b93/cv_bridge/CMakeLists.txt#L30) in `ros2_overlay_ws/src/vision_opencv/cv_bridge`
[image]

hariharan382, 2020-08-08:
please anyone help me to solve the error 
error while running"colcon build --symlink-install"