intel/ros2_openvino_toolkit #160: set_target_properties called with incorrect number of arguments

portgasray, 2019-11-22:
* branch: master
* ubuntu 18.04
* ros-dashing

**Output**:
```
--- stderr: dynamic_vino_lib                            
CMake Error at CMakeLists.txt:81 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

---
Failed   <<< dynamic_vino_lib   [ Exited with code 1 ]
```

RoboticsYY, 2019-11-22:
@LewisLiuPub This request comes from Fudan university robotics lab, could you please help him? Thanks.

portgasray, 2019-11-22:
I followed the instruction [OPEN_SOURCE_CODE_README]( https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md), and stop at last step build package in `ros2_overlay_ws`

RachelRen05, 2019-11-22:
@portgasray 
Can you check if there is a $CPU_EXTENSION_LIB variable in your environment?
This problem is mainly caused by the absence of the relevant variable of openvino, you followed the [OPEN_SOURCE_CODE_README](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md), so you should export the variable like this:
` export InferenceEngine_DIR=/opt/openvino_toolkit/dldt/inference-engine/build/`
` export CPU_EXTENSION_LIB=/opt/openvino_toolkit/dldt/inference-engine/bin/intel64/Release/lib/libcpu_extension.so`
` export GFLAGS_LIB=/opt/openvino_toolkit/dldt/inference-engine/bin/intel64/Release/lib/libgflags_nothreads.a`

portgasray, 2019-11-22:
> @portgasray
> Can you check if there is a $CPU_EXTENSION_LIB variable in your environment?
> This problem is mainly caused by the absence of the relevant variable of openvino, you followed the [...]

Thank you for your help, Although I put these `env`  in a file that convenient for me to `source`, I can hardly connect it with the problem here. 
As you say, I followed your suggestion and compiled the `ros2_overlay_ws` successfully (I had already deleted the `ros2_intel_realsense` part, due to the refactor branch has some problem to me)
Thank you all @RoboticsYY

LewisLiuPub, 2019-11-25:
Thank you, @RoboticsYY for bridging.
Thank you, @RachelRen05 for answering.
We are now in the progress of updating README files, in order to make them clearer and easier to understand.
I'd close this issue. and please feel free to reopen it if anyone has more comments.

closed 2019-11-25 by LewisLiuPub