intel/ros2_openvino_toolkit #162: I'm not able to run OpenVino toolkit

HartlD2, 2019-11-26:
I always get the following error if I want to install OpenVino ROS2 on Ubuntu 18.04 via environment_setup.sh:
Installing LibRealsense
gpg: can't open '/home/.../code/open_model_zoo/demos/build/C8B3A55A6F3EFCDE' : No such file or directory

If I install without librealsense, I will get the following error if I want to launch the package. 

An if I start: ros2 launch dynamic_vino_sample pipeline_people.launch.py

I will get the following error:
Package 'dynamic_vino_sample' not found: "package 'dynamic_vino_sample' not found, searching: ['/opt/ros/dashing']"

RachelRen05, 2019-12-09:
@HartlD2 
We have solved the librealsense installation problem, please refer to  pr #161 .
About the problem of "Package 'dynamic_vino_sample' not found", it's because you didn't source the OpenVino ROS2 environment. If you followed the installation [guide](https://github.com/intel/ros2_openvino_toolkit/blob/devel/doc/OPEN_SOURCE_CODE_README.md#4-building-and-installation), the command should be:
source ~/ros2_overlay_ws/install/local_setup.sh

jb-balaji, 2026-01-12:
Stale issue, hence closed, kindly create new issue if required.