intel/ros2_openvino_toolkit #164: dynamic_vino_sample use case

portgasray, 2019-11-27:
with `ros2 launch dynamic_vino_sample pipeline_segmentation.launch.py `
* the error information:
```
[ ERROR ] Error loading xmlfile: /opt/openvino_toolkit/models/segmentation/output/FP16/frozen_inference_graph.xml, File was not found at line: 1 pos: 0
```
[image]

the config file `ros2_openvino_toolkit/sample/param/pipeline_segmentation.yaml`

```
Pipelines:
- name: segmentation
  inputs: [RealSenseCameraTopic]
  infers:
    - name: ObjectSegmentation
      model: /opt/openvino_toolkit/models/segmentation/output/FP16/frozen_inference_graph.xml
      engine: "HETERO:CPU,GPU,MYRIAD"
      label: to/be/set/xxx.labels
      batch: 1
      confidence_threshold: 0.5
  outputs: [ImageWindow, RosTopic, RViz]
  connects:
    - left: RealSenseCameraTopic
      right: [ObjectSegmentation]
    - left: ObjectSegmentation
      right: [ImageWindow]
    - left: ObjectSegmentation
      right: [RosTopic]
    - left: ObjectSegmentation
      right: [RViz]

OpenvinoCommon:
```

RachelRen05, 2019-12-09:
@portgasray 
Did you download and convert a trained model to produce an IR,  refer to the segmentaton download and convert section in this [guide](https://github.com/intel/ros2_openvino_toolkit/blob/devel/doc/BINARY_VERSION_README.md#5-running-the-demo).

portgasray, 2019-12-14:
@RachelRen05 Hi, I need help. I can not exec `python3 ./downloader.py --name mobilenet-ssd
` in [Object Detection](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OBJECT_DETECTION.md), it needs to connect to the host `docs.google.com`, any suggestion?

RachelRen05, 2019-12-16:
@portgasray 
Hi, we have never encountered this problem, can you provide a detailed error log?

portgasray, 2019-12-20:
@RachelRen05 
Sure, first. the [guide ](https://github.com/intel/ros2_openvino_toolkit/blob/devel/doc/BINARY_VERSION_README.md#5-running-the-demo) you provided I could not use it (404 page). 
then I found the new [guide ](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/BINARY_VERSION_README.md)according to the URL string `BINARY_VERSION_README`.

I perform the `5. Running the Demo` under the guide of [BINARY_VERSION_README](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/BINARY_VERSION_README.md).

It is also a 404 page when I clicked `download Object Detection model ` in the second step of `Preparation`.(See the following)
[image]

I tracked the page where it jumped to and found the [OBJECT_DETECTION](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/OBJECT_DETECTION.md) instead.
[image]
[image]

Finally, My issue happened here, when I perform the step `mobilenet-ssd`,  the second command `python3 ./downloader.py --name mobilenet-ssd` occored the following error:
```
################|| Downloading models ||################

========== Downloading /home/lab001/workspace/sofware_build/open_model_zoo/tools/downloader/public/mobilenet-ssd/mobilenet-ssd.prototxt
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 83, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 601, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 346, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 852, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 284, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x7fbf26b16780>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 639, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7fbf26b16780>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./downloader.py", line 65, in try_download
    chunk_iterable = start_download()
  File "./downloader.py", line 244, in <lambda>
    lambda: model_file.source.start_download(session, CHUNK_SIZE)):
  File "/home/lab001/workspace/sofware_build/open_model_zoo/tools/downloader/common.py", line 172, in start_download
    response = session.get(self.url, stream=True, timeout=DOWNLOAD_TIMEOUT)
  File "/usr/local/lib/python3.6/dist-packages/requests/sessions.py", line 546, in get
    return self.request('GET', url, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.6/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7fbf26b16780>: Failed to establish a new connection: [Errno 111] Connection refused',))
########## Error: Download failed

################|| Post-processing ||################

FAILED:
mobilenet-ssd
```
In my opinion, It caused by the network connection.

portgasray, 2019-12-20:
```
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt)
```
according to the error info, I performed an experiment that I combined the [url](https://raw.githubusercontent.com/chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt).
`https://raw.githubusercontent.com/chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt`
First, I opened on the web browse and it succeed.
[image]

then I used the `wget `on the server where the error happened
```
$ wget https://raw.githubusercontent.com/chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt
--2019-12-20 12:32:58--  https://raw.githubusercontent.com/chuanqi305/MobileNet-SSD/ba00fc987b3eb0ba87bb99e89bf0298a2fd10765/MobileNetSSD_deploy.prototxt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.228.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.228.133|:443... failed: Connection refused
```

portgasray, 2019-12-20:
finally, I download two files manually!
[image]

closed 2023-04-24 by LewisLiuPub