intel/ros2_openvino_toolkit #179: Mixed ros crystal and  dashing version in environment_setup.sh on dashing branch

chuan9, 2019-12-24:
for dashing branch, 109 line in environment_setup.sh is as bellows:
```
 rosdep install --from-paths src --ignore-src --rosdistro crystal -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers"
```

Should we install dashing version here?

LewisLiuPub, 2019-12-24:
Hi, it's a typo. you needn't install dashing version here. suppose Eloquent is supported by this branch.

chuan9, 2019-12-24:
@LewisLiuPub Hi, the 95 line in the same file:
```
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
```
It actually doesn't download dashing verison here also.
BTW, I want to use it on eqolent, but other intel git repos, such as [realsense](https://github.com/intel/ros2_intel_realsense),[ros2_object_analytics](https://github.com/intel/ros2_object_analytics/tree/eloquent) , are based on dashing. So for the consistency, I need to use dashing version.

LewisLiuPub, 2019-12-27:
@chuan9 yes, I also suggest that you try dashing first, since eloquent is too new to be stable.

BTW, for the Intel's ROS2 projects, you can follow below tables for a quick trial with eloquent branches, if you are eager to use new features in branch eloquent.

Project/Purpose | Regular development | Dashing hot fix | Eloquent (enabling and hot fix)
-- | -- | -- | --
Ros2_intel_realsense | refactor | dashing | eloquent
Ros2_openvino_toolkit | devel | dashing | eloquent
Ros2_object_analytics | devel | master | eloquent
Ros2_grasp_library | master | master | master
Navigation2 | master | dashing-devel | eloquent-devel (to be created, and hot fix only)
Robot_devkit | devel | N/A | eloquent (enabling only)

closed 2023-04-24 by LewisLiuPub