intel/ros2_openvino_toolkit #189: Suggested OpenVino version ?

cwyark, 2020-10-27:
Is there any suggested openvino version ? 
I download binary openvino from website (openvino_2020.3.194) and trying to use that with ros2_opevino_tooklit. 
I got an error wher env CPU_EXTENSION_LIB is not set. But I could not found any libcpu_extension.dylib under the binary. 

PS: I use OSX. v10.14.6

LewisLiuPub, 2020-10-27:
* It is probably because of the current verison doesn't support OpenVINO_2020.3.xxx. We are now in the progress of upgrading this package to support it in [a private repo](https://github.com/LewisLiuPub/ros2_openvino_toolkit/tree/dev-ov.2020.3) , in which object detection is enabled. It is highly recommended to use this repo for v2020.3 attemps.
* We didn't test this package under OSX. If possibly, please try to round it under Ubuntu system to be familiar with the build environment.

cwyark, 2020-10-27:
Thanks for clarification.

closed 2023-04-24 by LewisLiuPub