intel/ros2_openvino_toolkit #192: New development features in Y2021?

LewisLiuPub, 2020-12-15:
Hi, ROSers,

I am now planning the new features and development plans for ROS/ROS2 OpenVINO Toolkit projects.
In order to better fit the feature requirements with the limited work load, I would like to gather requirements from the developers, real users or field experts as the first step, then filter them in a feasible development priorities and schedules.

Here are some thoughts about migration/development activities, wish anyone interested can share yours:

* Support latest ROS 2 versions and OpenVINO versions
* Support more popular models and inferences, to better fix robotics use scenarios
* GUI-based  AI Inference Pipeline Configuration tool, to ease the OpenVINO Toolkit's use and deployment.
* Docker image support

Please feel free to share your thoughts, what features do you want this project have?

SteveMacenski, 2020-12-15:
Naming the specific models available, not just the general types of things. It matters to professional users not just image detection, but specifically what network trained on what dataset and how to deploy retrained networks on custom datasets. 

Having instance segmentation models like mask rcnn or yolact would be greatly beneficial to the work in nav2 for non-Nvidia users. 

Using `vision_msgs` would be good since it generalizes most of your custom messages away. If `vision_msgs` is missing something you need, file a ticket and lets make it happen. 

List some metrics from the networks you do support (what resolution of images at what fps can you achieve with each using this?)

joelbudu, 2021-02-27:
Dear @LewisLiuPub,
Thank you for the opportunity and the great work you have been doing here on this repository. I find it extremely useful in making deployments with openvino in ros.
What would be really useful as you have mentioned is to : **Support more popular models and inferences, to better fix robotics use scenarios**
I'm particularly interested in having model support and a pipeline for semantic segmentation models like Deeplabv3 and also support for more models in object detection like Faster RCNN and Yolo3,4 

Hope to hear updates on these soon. Thanks

LewisLiuPub, 2021-03-01:
Thank you very much for your inputs, @SteveMacenski @joelbudu 
Really agree that the features (new modules, new interfaces) are important for robot AI scenarios. I'll put efforts to think of their implementation, and also appreciate any contribution from the community to this project with any development.

closed 2023-04-24 by LewisLiuPub