intel/ros2_openvino_toolkit #198: Preferred version of Ubuntu

sapoluri, 2021-02-12:
The ROS2 version I intend to use is Foxy. Foxy's preferred Ubuntu version is 20.04.

OpenVino 2021.1/2021.2 both support Ubuntu 20.04. Will ros2_openvino_toolkit work with OpenVino 2021.1/2021.2?

LewisLiuPub, 2021-02-12:
yes, in progress.

sapoluri, 2021-02-12:
Any estimate on when it might released?

LewisLiuPub, 2021-03-24:
I am now in progress of porting this project to support Ubuntu 20.04.
Could you please let me know more about your use of this project, thus I can judge correctly and make a better development schedule for you? @sapoluri

LewisLiuPub, 2021-04-28:
@sapoluri 
Ubuntu20.04 has been officially supported by now. please track the new branch: [dev-ov.2021.3](https://github.com/intel/ros2_openvino_toolkit/tree/dev-ov.2021.3)

closed 2021-04-28 by LewisLiuPub