intel/ros2_openvino_toolkit #200: double free or corruption (out)  with Yolov2 Object detection

joelbudu, 2021-02-15:
I am trying to deploy a custom trained YoloV2 model for the object detection pipeline in the dynamic_vino_lib samples. However I am getting a double free or corruption (out)
Aborted (core dumped) when I launch the object detection node. Can't tell what the problem is because the log doesn't give any clue. Can you help?

```
[ INFO ] Loading network files
[ INFO ] Batch size is set to  1
[ INFO ] Checking input size
[ INFO ] Checking output size
[ INFO ] Checking Object Detection outputs
[ INFO ] Checking Object Detection outputs ...
[ INFO ] Checking Object Detection output ... Name=output/YoloRegion
[ INFO ] Checking Object Detection num_classes
[ INFO ] Checking Object Detection output ... num_classes=3
[ INFO ] max proposal count is: 110
[ INFO ]  ... Adding one Inference: ObjectDetection
[ INFO ] Updating connections ...
[ INFO ] Checking connection into pipeline:[ObjectDetection(2)<-->RosTopic(3)]
[ INFO ] Adding connection into pipeline:[ObjectDetection<-->RosTopic]
[ INFO ] Checking connection into pipeline:[RealSenseCameraTopic(1)<-->ObjectDetection(2)]
[ INFO ] Adding connection into pipeline:[RealSenseCameraTopic<-->ObjectDetection]
[ INFO ] One Pipeline Created!
 --> RealSenseCameraTopic
ObjectDetection --> RosTopic
RealSenseCameraTopic --> ObjectDetection
double free or corruption (out)
Aborted (core dumped)

```

LewisLiuPub, 2021-03-08:
Thank you very much for your reports.
Which branch are you using? which OpenVINO toolkit version(2020.3?) are you depdent on?
Please let me know more about your SW configuraiton and env settings when this error occurred.

joelbudu, 2021-03-18:
@LewisLiuPub It was on OpenVINO toolkit version 2019R3.1 

I realized my mistake, My network input size was not square (1280 x 736), it runs fine with square input sizes.

closed 2021-03-18 by joelbudu

LewisLiuPub, 2021-03-19:
Thank you very much for your inputs. it sounds that there are some important info about module properties in this project's README, I am thinking of adding this kind of info later on. Thanks.