intel/ros2_openvino_toolkit #201: Parallel acceleration for detection models

joelbudu, 2021-03-18:
@LewisLiuPub  is possible to get the functionality to have parallel execution for detection models as similar to this PR? - #143. Would be very useful to get faster speeds especially on the Movidius VPU

LewisLiuPub, 2021-03-24:
Hi, @joelbudu 
Do you need to input multiple data from several cameras? or only use one camera but want to execute the inference by several Movidius VPUs in parallel?

joelbudu, 2021-03-24:
@LewisLiuPub I would like to use one camera but execute the inference by several Movidius VPUs in parallel, please.

LewisLiuPub, 2021-03-24:
@joelbudu Can you let me know more, which model(s) are you using in this case?

joelbudu, 2021-03-24:
Object Detection -> YOLOV2
Object Segmentation -> MaskRCNN

LewisLiuPub, 2023-04-26:
To update the status for multiple pipeline supporting and yolo/maskrcnn supporting:
- Yolov2 is deprecated. Instead, Yolov5 is supported. 
- instance segmentation (maskrcnn, deeplabv3) is supported on branch ros2 (enabled for ros2 foxy/galactic)

closed 2023-04-26 by LewisLiuPub

joelbudu, 2023-04-26:
Thank you for working on this @LewisLiuPub