intel/ros2_openvino_toolkit #255: Object detection yolo v4 support

hcdiekmann, 2022-12-09:
The default supplied object detection model ([ssd_mobilenet_v2_coco](https://docs.openvino.ai/2021.4/omz_models_model_ssd_mobilenet_v2_coco.html)) has a bad precision of >25% mAP.
However after testing the object detection node with multiple newer models converted to IR format, for example yolo-v4-tf, it seems that models with more than one output layer are not supported and cause an error.

LewisLiuPub, 2023-04-26:
Recently there are several requirements from real user/customers for supporting Yolo v7. sounds now it is the good time to enable it.

Get the model info from [the openvino nootbooks](https://github.com/openvinotoolkit/openvino_notebooks/tree/main/notebooks/226-yolov7-optimization).

The possible steps to complete the enabling task:
- Get the [YOLOV7 Pytorch pre-trained files](https://github.com/WongKinYiu/yolov7).
- Reference [the steps](https://github.com/intel/ros2_openvino_toolkit/blob/master/doc/quick_start/tutorial_for_yolov5_converted.md) to convert  yolov5 model to OpenVINO optimized model.
- Either: Try to update model class for Yolov5 to support both Yolov5 and yolov7.
- Or if the existing class for yolov5 model doesn't support yolov7: 
    - Add new sub-class as child of [base model](https://github.com/intel/ros2_openvino_toolkit/blob/master/openvino_wrapper_lib/include/openvino_wrapper_lib/models/base_model.hpp) for Yolov7 model parsing.
    - Update [Output ImageWindow](https://github.com/intel/ros2_openvino_toolkit/blob/master/openvino_wrapper_lib/include/openvino_wrapper_lib/outputs/image_window_output.hpp) to support the new model.

LewisLiuPub, 2025-02-20:
fixed.

PR #290 (2023-05-19): [Issue: #255] Add YOLO v7 support
Add support for YOLO v7
Add new class for YOLO v7 model
Modify PipelineManager to run either v5 or v7 model 
Add documentation to explain YOLO v7 model conversion

Fixes #255

---------
commit 122144de (2023-05-19): [Issue: #255] Add YOLO v7 support (#290)

Add support for YOLO v7
Add new class for YOLO v7 model
Modify PipelineManager to run either v5 or v7 model
Add documentation to explain YOLO v7 model conversion

Fixes #255

---------
closed 2025-02-20 by LewisLiuPub