intel/ros2_openvino_toolkit #286: Refine matToBlob function from derived classes to BaseModel

gitpqLee, 2023-04-24:
Currently, in [ObjectDetectionModel ](https://github.com/intel/ros2_openvino_toolkit/blob/master/openvino_wrapper_lib/include/openvino_wrapper_lib/models/base_model.hpp#L138), the virtual function matToBlob has no implementation and each derived classes have their own implemantation. 
We want to refine the common function to the ObjectDetectionModel or BaseModel. For the model specific pre-process, we can use some functions to implemente which can be overrideed. For example:
bool matToBlob (argc, argv*) {
      [some common implementation];
      func_1();
      func_2();
}
func_1() / func_2() can be overrideed by derived model classes and implement some model specific pre-process logic.

ChloeWu1, 2023-04-25:
ROS_Team2 working on this issue.

ChloeWu1, 2023-04-28:
Please review #289

closed 2025-02-20 by LewisLiuPub