intel/ros2_openvino_toolkit #306: Object Detection Yolov8 Support

Zephyr-S, 2023-09-18:
I'm trying to replace the ssd_mobilenet_v2_coco object detection model with my YOLOv8 model on the AMR. I have already modified the paths and related files (xml, label, bin). But when running the object detection docker, the model path did not change. The error shows as below:

```
[setupvars.sh] OpenVINO environment initialized
[INFO] [1694965557.795466727] [i045.object_detection_node]: [ObjectDetection] Getting parameters ...
[INFO] [1694965557.795635151] [i045.object_detection_node]: Set model_path to: /home/eiforamr/workspace/object_detection/src/prebuilt_object_detection/object_detection/share/object_detection/models/ssd_mobilenet_v2_coco/frozen_inference_graph.xml

[INFO] [1694965557.795653094] [i045.object_detection_node]: Set device to: CPU

[INFO] [1694965557.795666811] [i045.object_detection_node]: Publishing to /camera/detected_objects
[INFO] [1694965557.798381064] [i045.object_detection_node]: Publishing to /camera/image_tracked
[INFO] [1694965557.800236433] [i045.object_detection_node]: Publishing to /camera/tracked_objects
[INFO] [1694965557.800437660] [i045.object_detection_node]: Publishing to /object/image_for_remote
[INFO] [1694965557.802707843] [i045.object_detection_node]: Subscribing to /camera/color/image_raw
[INFO] [1694965557.803233690] [i045.object_detection_node]: Subscribing to /camera/aligned_depth_to_color/image_raw
[INFO] [1694965557.803447426] [i045.object_detection_node]: Subscribing to /camera/aligned_depth_to_color/camera_info
[INFO] [1694965557.803859974] [i045.object_detection_node]: Subscribing to /object/remote_results
[INFO] [1694965557.804077811] [i045.object_detection_node]: Subscribing to /object/enable_remote_inference
InferenceEngine: 0x7fab884aa030
Loading Inference Engine
Loading network files
Path to model: /home/eiforamr/workspace/object_detection/src/prebuilt_object_detection/object_detection/share/object_detection/models/ssd_mobilenet_v2_coco
Model file name: frozen_inference_graph.xml
Base file name: frozen_inference_graph
[ INFO ] Loading Inference Engine
[ INFO ] Device info: 
[ INFO ] 	CPU
	MKLDNNPlugin version ......... 2021.4
	Build ........... 2
Loading network files
[ ERROR ] Unknown model format! Cannot find reader for model format: xml and read the model: /home/eiforamr/workspace/object_detection/src/prebuilt_object_detection/object_detection/share/object_detection/models/ssd_mobilenet_v2_coco/frozen_inference_graph.xml. Please check that reader library exists in your PATH.
[INFO] [1694965557.845248517] [i045.object_detection_node]: Start Tracking Bounding Boxes
[WARN] [1694965558.880644127] [i045.object_detection_node]: No data received in CameraTopic instance
```

I tried to search the "ssd_mobilenet_v2" in the folder and the result is:

```bash
$ grep -r -n ssd_mobilenet workspace
Binary file workspace/object_detection/src/prebuilt_object_detection/object_detection/bin/object_detection_node matches
Binary file workspace/object_detection/src/prebuilt_object_detection/object_detection
```
I can find the model_path in the two files:
[image]
How can I compile my own Yolov8 model and update the model path?

LewisLiuPub, 2024-07-26:
This issue happened 2 errors:
- [Error-1] You probably didn't set `mode_type` as `yolov8` in the config file, please see this [config file](https://github.com/intel/ros2_openvino_toolkit/blob/master/sample/param/pipeline_object_yolo.yaml) as reference.
-  [Error-2] The implementation in the repo doesn't match the latest [ultralytics](https://github.com/ultralytics/ultralytics) pretrained yolov8.2 model, which should be fixed.

So this issue is still openning for `Error-2`: Update related code to support latest ultralytics models:
- Yolov8 ( Detection & Segmentation)
- Yolov10 (Detection)
- Others (Optional)

LewisLiuPub, 2025-02-20:
Yolov8  has already been supported.
Yolov10/11 is still open for supporting.

LewisLiuPub, 2025-02-21:
Create new issues (#318 , #319 ) for Yolov11 support . This issue is closed.

closed 2025-02-21 by LewisLiuPub