intel/ros2_openvino_toolkit #310: New target device support (MOVIDIUS/VPU)

RodBelaFarin, 2024-02-20:
After a very long and painful installation procedure, I have successfully installed the ROS2 OpenVINO toolkit for Humble on Ubuntu 22.04 with OpenVINO runtime 2023.3.0.

Currently it is working with CPU, but I was expecting to be able to use my Intel Movidius MyriadX.

Can you add this feature, please?

nilutpolkashyap, 2024-03-20:
Hi @RodBelaFarin , Openvino 2023.3.0 no longer support  Intel® Movidius ™ VPU based products. The support for these were dropped from the 2022.3.0 release onwards.

[https://community.intel.com/t5/Intel-Distribution-of-OpenVINO/MYRIAD-support-for-openvino-2022-3/td-p/1446425](https://community.intel.com/t5/Intel-Distribution-of-OpenVINO/MYRIAD-support-for-openvino-2022-3/td-p/1446425)

[https://github.com/openvinotoolkit/openvino/issues/15537](https://github.com/openvinotoolkit/openvino/issues/15537)

Thanks

hance02, 2024-11-23:
Hi @RodBelaFarin ，This issue has been troubling me for a while. Can you help me with how to install the ROS2 OpenVINO Toolkit on Ubuntu 22.04 and use ROS2 Humble to run OpenVINO Runtime 2023.3.0?